simbody/simbody #758: Setting output precision for String::String(const T& t)

fcanderson, 2023-08-26:
@sherm1 
I hope you are doing well.  I took a break from coding for a few months, but I'm back to it now.  I have few minor tweaks that I'd like to make to Simbody.  Before I submit a PR, I'd like to check with you to see how you'd like me to go about that. 

First up are some almost-one-liner changes I'd like to make in String.h and in Xml.h.

------
Some background
------
In OpenSim, I have implemented class StatesDocument, which follows through on the planned roadmap capability of serializing ALL states (continuous, regular discrete, auto-update discrete, modeling option). Currently, OpenSim employs no systematic approach for serializing/deserializing discrete states. When they are encountered (e.g., as the input controls for a simulation), they are handled in an ad hoc way (e.g., using a separate file with its own format to store the discrete states).  I ran into trouble when bringing my exponential spring contact class into OpenSim, as class ExponentialSpringForce uses discrete states (i.e., the coefficients of friction, elastic anchor point, and sliding state).  Class StatesDocument will allow all states to be serialized systematically in a way that accommodates a wide range of variable types and that simplifies implementation. Ultimately, I think this will also reduce the complexity that end-users will need to deal with (e.g., to reproduce simulated states, just one .ostates file will be needed). 

I have used class SimTK::Xml as the basis for serialization/deserialization in class OpenSim::StatesDocument.  StatesDocument fell out quite naturally as a result. The class is up and running. There was only one feature that I couldn't figure out how to specify without a code change in Simbody:  **output precision**.  That is, how many significant figures are written to file.  To exactly reproduce a simulation, we'd need lossless serialization/deserialization.  However, for some situations where loss-less deserialization is not needed, reducing file size could be helpful.

Currently, the call chain in Simbody preceding serialization for the case of an ```Array_<Vec3>``` (to make things concrete) is:
```
// In calling code
Array_<Vec3> someValue;   \\ initialized somehow
someXmlElement.setValueAs<Array_<Vec3>>(someValue);

--> // In class SimTK::Xml::Element
    void setValueAs(const Array_<Vec3>& value) {
        setValue(String(value));
    }

    --> // In class SimTK::String in the overload constructor
        String::String(const Array_<Vec3>& t) {
            std::ostringstream os;
            *this = stringStreamInsertHelper(os, t, true).str();
        }
```
Here, the creation of the string that will ultimately be written to file has the default precision, which is 6, at least on Windows in Visual Studio 19.  There may be some way to change the global default for output precision.  Even if so, however, I think taking steps to alter global output precision and then return it to some agreed upon baseline value would be messy. 

------
Changes I'd like to merge.
------
I'd like to add a `precision` argument to both Element::setValueAs(const T& value) and to String::String(const T& value) that has a default value of ```SimTK::LosslessNumDigitsReal```.  So, the modified code would be

```
// In calling code
int precision = 8;   \\ or whatever is determined appropriate by the caller.
Array_<Vec3> someValue;   \\ initialized somehow
someXmlElement.setValueAs<Array_<Vec3>>(someValue, precision);
// Alternatively, the following call without the precision argument produces lossless serialization
// someXmlElement.setValueAs<Array_<Vec3>>(someValue);

--> // In class SimTK::Xml::Element
    void setValueAs(const Array_<Vec3>& value, int precision=SimTK::LosslessNumDigitsReal) {
        setValue(String(value), precision);
    }

    --> // In class SimTK::String in the overload constructor
        String::String(const Array_<Vec3>& t, int precision=SimTK::LosslessNumDigitsReal) {
            std::ostringstream os;
            os << std::setprecision(precision);
            *this = stringStreamInsertHelper(os, t, true).str();
        }
```
A nice feature of the above approach is that the output precision is modified locally.  As soon as the call chain completes, the output precision is back to whatever the global default is.

I have implemented the above changes.  All existing Simbody regression tests pass.  In OpenSim, class StatesDocument is functioning as intended.  The control of output precision has been straight forward.  

------
 Issues / Questions
------
1. First, is there already a way to control output precision in Simbody for objects that implement the ```<<``` operator that I did not find but should be using? 
2. Default Precision. Should the default serialization precision be ```SimTK::LosslessNumDigitsReal```? Choosing this increased precision could alter the output that folks get. When debugging, for example, they could get 20 digits for each number, instead of only 6. I note here that the construction of Strings via other overloaded constructors appears to make the same choice.  That is, in the absence of a specified output format, max precision is what is delivered.
3. Testing.  Should I enhance regression testing for classes SimTK::String and SimTK::Xml to focus on output precision?  Or, are these changes basic enough that additional tests are not needed? <**bowing head in prayer that additional tests are not needed**>
4. Trailing White Space.  Both String.h and Xml.h have many lines with trailing white spaces.  Should I first submit a PR with the trailing white spaces removed so that when I submit the substantive PR the output precision related changes will be easy to find and not camouflaged by zillions of incidental edits? I could leave the trailing white spaces, but that would actually be more of a pain because in my versions of the files the whitespace has already been removed.

I usually feel apologetic for the long messages I send, but my inclination to be thorough invariably wins out. I'm hoping the length here will ultimately require less total time from you and maybe others.

Thanks,
Clay

sherm1, 2023-08-28:
Hi, Clay! This seems like a good fix, thanks. My first thought is -- why not just change the output to lossless and _not_ add a precision parameter? If there is no immediate need for reduced precision, let's not add the extra complexity. It could always be added later if a necessary use case is discovered. I don't think reduced file size is a compelling need.

With no additional parameter, I don't think a new unit test would be required. With an API change, I do think a small test would be required to ensure that it works and keeps working in the future.

Regards, Sherm

fcanderson, 2023-08-28:
Sherm, thanks for the response.  What you suggest { just changing the output to lossless always } would be simplest.  And, as you say, it can change later pretty simply if a need arises.

Do you care about how I deal with the trailing whitespace in String.h?

I am inclined to do two PRs:
PR1 to get rid of the trailing whitespace in String.h (and perhaps also Xml.h), and
PR2 to make the output precision change to lossless.

sherm1, 2023-08-28:
Two PRs would be ideal, thanks.

fcanderson, 2024-03-05:
#776 and #779 add the capacity to convert numerical types to ```SimTK::String```s and ```Xml::Element``` values.

closed 2024-03-05 by fcanderson