simbody/simbody #764: "No such file or directory"

MuhammadUtoro, 2023-09-10:
Hi everyone, I was trying to build simbody from the source code for OpenSim purpose. I followed every step from the Github page and it seems working fine. However, as I would test the pendulum code in the visual studio this error always pops out. I have searched for solution (inlclude/lib, etc.) and they cant' seem work.

[image]

I am very keen to make this work, therefore I tried to create a solution file in visual studio and it worked! My question is, does this mean that I cannot use Simbody with vscode? Or did I miss any important steps?
Thank you for your help

sherm1, 2023-09-11:
Looks like it couldn’t find the header file “Simbody.h”. Likely vscode just needs to be told where to look?

 

 <https://user-images.githubusercontent.com/115923671/266848580-90df0cb4-b197-4271-9051-4801818b1b23.jpg>

sherm1, 2023-09-11:
Unfortunately I'm not a vscode user so I don't know the magic incantation. Maybe someone else here can advise?

MuhammadUtoro, 2023-10-26:
Hi, could anybody help me out to solve this problem? I have tried different ways to find out the problem with this "No such file or directory" issue? I have tried to build and install with CMake and also installing from conda-forge, this issue always pops out. Is it not possible to use simbody with visual studio code? 
I appreciate your help

aymanhab, 2023-10-26:
While I do not develop on vscode, it looks like there's a plugin to use CMake in vscode https://devblogs.microsoft.com/cppblog/cmake-tools-extension-for-visual-studio-code/
This is more a question to the CMake/vscode community rather than simbody but let us know hwat you fins out as would like to document the steps since more users are likely to use vscode moving forward. Thank you

MuhammadUtoro, 2023-10-30:
> While I do not develop on vscode, it looks like there's a plugin to use CMake in vscode https://devblogs.microsoft.com/cppblog/cmake-tools-extension-for-visual-studio-code/ This is more a question [...]
Hi, thanks for your answer. I am not sure about the link you provided, but at least it is working now for me in vscode. Appreciate it!