simbody/simbody #765: Visualizer crashes when drive letters don't match

mrrezaie, 2023-11-01:
Hi, I'm a Windows 10 with OpenSim v4.4.1 user and when the **Python current working directory** and the **simbody-visualizer executable directory** have different drive letter, the console crashes.

But when the drive letters are identical, it works well:
[image]

sherm1, 2023-11-01:
Good find! Anyone interested in tracking this down and submitting a PR to fix it?

@mrrezaie you might want to post this to OpenSim as well -- not clear from your description how the visualizer got invoked or what exactly crashed.

mrrezaie, 2023-11-01:
@sherm1 thank you. Please see:
https://github.com/opensim-org/opensim-core/issues/3344#issuecomment-1775421023
https://github.com/opensim-org/opensim-core/issues/3344#issuecomment-1780203065

sherm1, 2023-11-02:
Is the problem that some DLL the visualizer depends on isn't in a place Windows can find it?

mrrezaie, 2023-11-03:
I don't think so. In that case, I would have got these errors:
[image]
[image]

mrrezaie, 2024-04-27:
@sherm1, Hi again, I'm pasting this error for your record (visualizer error when the drive letters differ). 

The Python script is located in drive `D:\`, the OpenSim executable files (Python installation folder) in drive `F:\`, and the windows files in drive `C:\`.

Please also note that the last three paths as well as the directories end with `\C\` or `\F\` do not exist in my system. 

Perhaps the issue is the way the directories are split. For example, the third path where the OpenSim executables exist should become `F:\Python38\Lib\site-packages\opensim\` (starts with `F` instead of `d`)

## Error

`RuntimeError: std::exception in 'void OpenSim::MocoStudy::visualize(OpenSim::MocoTrajectory const &) const': SimTK Exception thrown at VisualizerProtocol.cpp:167:`
`  Error detected by Simbody method VisualizerProtocol::ctor(): Unable to spawn executable 'simbody-visualizer' from directories:`
`  f:\Python38\`
`  d:\F\`
`  d:\Python38\Lib\site-packages\opensim\`
`  d:\C\`
`  d:\Windows\system32\`
`  d:\C\`
`  d:\Windows\`
`  d:\C\`
`  d:\Windows\System32\Wbem\`
`  d:\C\`
`  d:\Windows\System32\WindowsPowerShell\v1.0\`
`  d:\C\`
`  d:\Windows\System32\OpenSSH\`
`  d:\F\`
`  d:\Python38\`
`  d:\F\`
`  d:\Python38\Scripts\`
`  d:\C\`
`  d:\Users\mrr\AppData\Local\Microsoft\WindowsApps\`
`  d:\C\`
`  d:\Users\mrr\AppData\Local\GitHubDesktop\bin\`
`  C:/Users/runneradmin/opensim_dependencies_install/simbody/bin/`
`  C:\Program Files\Simbody\bin\`
`  C:\Program Files\SimTK\bin\`
`Final system error was errno=2 (No such file or directory).`
`  (Required condition 'status == 0' was not met.)`

Thanks for your time.

mrrezaie, 2024-05-06:
Hi @sherm1, I think a temporary fix would be to use `getThisExecutableDirectory` instead of `getCurrentWorkingDirectory` in following line; this explain why an incorrect drive letter was combined with the opensim dll/executables path.
https://github.com/simbody/simbody/blob/d98f05fa1899633b544757e6c179562940f0849d/SimTKcommon/include/SimTKcommon/internal/Pathname.h#L180

But still this question remains why the drive letters were separated from the corresponding directories (green), and another one was combined with them (red). 
[image]

mrrezaie, 2024-05-23:
Hi @sherm1, @aymanhab, @nickbianco 
Sorry for reaching out to you here. As can be seen in the last two comments, this issue is only related to the `simbody-visualizer.exe` path, and I think it won't take too much to be resolved. 
Just wanted to bring this to your attention for the new official release.
Sorry again, looking forward to the fix. Thank you in advance.