simbody/simbody #768: `getContiguousScalarData` has different behavior for `RowVector` and `RowVectorView`

nickbianco, 2023-11-10:
When accessing rows from a `Matrix`,  `RowVector`s and `RowVectorView`s return different values when accessing element via  `getContiguousScalarData`. It appears that `RowVectorView` is accessing data in the wrong order, whereas constructing a `RowVector` from the same row produces the expected behavior. This issue appears to only apply to `RowVectorView` and not `VectorView`.

### To reproduce
The code snippet,
```c++
SimTK::Matrix m(3, 2);
m.set(0, 0, 1.0);
m.set(1, 0, 2.0);
m.set(2, 0, 3.0);
m.set(0, 1, 4.0);
m.set(1, 1, 5.0);
m.set(2, 1, 6.0);

SimTK::RowVectorView rowview = m.row(0);
SimTK::RowVector row = m.row(0).getAsRowVector();

std::cout << "RowVectorView: " << std::endl;
std::cout << rowview.getContiguousScalarData()[0] << std::endl;
std::cout << rowview.getContiguousScalarData()[1] << std::endl;
std::cout << rowview.getContiguousScalarData()[2] << std::endl; // should be out-of-range
std::cout << std::endl;

std::cout << "RowVector: " << std::endl;
std::cout << row.getContiguousScalarData()[0] << std::endl;
std::cout << row.getContiguousScalarData()[1] << std::endl;
std::cout << row.getContiguousScalarData()[2] << std::endl; // is out-of-range
```
produces the following output:
```
RowVectorView:
1
2
3

RowVector:
1
4
0
```

### Context
We use `getContiguousScalarData` in OpenSim to convert rows (and columns) of a `TimeSeriesTable` [to numpy arrays](https://github.com/opensim-org/opensim-core/blob/7703592938979b54dc78ef4561781bbb7cc4994a/Bindings/Python/swig/python_simbody.i#L226). Since rows from a `TimeSeriesTable` are returned (e.g., from `getRowAtIndex`) as `RowVectorView`s, if they are converted to Python arrays they will contain the wrong elements.

sherm1, 2023-11-11:
Yikes! Great find, Nick -- I'll take a look.

sherm1, 2023-11-11:
OK, I took a closer look. This is working correctly. There are several oddities in the example above. In a Debug build the example would have failed because getContiguousScalarData() can only be called when the data is actually contiguous! (You can check with hasContiguousData()).

Recall that Simbody stores matrices in column order. So your 3x2 matrix above, which logically is
```
1 4
2 5
3 6
```
is stored in memory as `1 2 3 4 5 6`.

Note that your rows are only 2 elements long. So it isn't clear to me why you were looking at 3 elements above, what you were expecting, or why you were satisfied with the RowVector output.

A "view" is a look into the original data. So RowVectorView of row 0 points to the first element but has a stride of 3 so that it goes to the top of the next column (that's what's meant by "not contiguous data"). OTOH, when you make a copy of the row into a new RowVector, the copy is stored in contiguous data which is why you got the expect 1,4 when you looked at the contiguous data. The final "0" in that case is just what happened to be in memory since there were only two legitimate elements there. For the RowVectorView, in Debug mode it asserts hasContiguousData() which would have failed, but in Release it just returns a pointer to the first element. Then when you printed out 3 elements you got the 1 2 3 of the first column because of the memory layout.

LMK if you have further questions.

closed 2023-11-11 by sherm1

nickbianco, 2023-11-13:
Thanks @sherm1, that all makes sense! I suspected it was a memory storage thing, but wasn't sure (hence why I was trying to access elements outside the row to see if it actually went out-of-bounds).