simbody/simbody #775: Variable precision for String::String(const T& t) and Xml::setValueAs(const T& t)

fcanderson, 2024-01-30:
@sherm1 I hope you are well! It's been a while.

I'm following up about setting the output precision in ```String::String(constT& t)``` to lossless.  See #762.

Based on interacting with the OpenSim Developer Community at its monthly meetings, there is consensus that the ability to vary output precision would be desirable.  Fixing the precision to lossless impacts both OpenSim model files (.osim files) and state serialization files (.ostates files), both of which use SimTK::Xml to write the files.

For model files, a precision of ~6 is preferred.  Writing elements with lossless precision (20) can make the model files more challenging to read and edit.  (User still manually edit model files.)  In addition, situations arise where a number like 1.5 is changed to 1.499999999999999999999.  Ayman made the comment that such changes make displaying numbers in the GUI an issue and file diffs become harder to interpret.  Ajay made the comment that we don't want to give OpenSim users the impression that the values of model parameters are known to a precision of 20.

For state serialization, the desired precision could vary.  For plotting, a precision ~6 is probably more than enough, while other applications (e.g., perturbation analysis, resuming a simulation midway through, etc.) might require lossless precision.

Would you be open to me submitting a PR that adds variable precision? I propose the following changes to the SimTK API:

```String::String(const T& t)```    -->  ```String::String(const T& t, int precision = 6)```
```Xml::Element::setValueAs(const T& value)```  -->  ```Xml::Element::setValueAs(const T& value, precision = 6)```

6 is the default C++ output precision. It's what users have unknowingly been using.

I would be happy to augment current unit tests and update the documentation.

Thanks,
Clay

sherm1, 2024-01-30:
Sounds like a good plan!

fcanderson, 2024-01-31:
Great.  I'll probably submit the PR in about a week.  Thanks.

fcanderson, 2024-03-05:
#776 and #779 add the capacity to specify numerical precision for conversion to ```SimTK::String```s and XML document elements.

closed 2024-03-05 by fcanderson