simbody/simbody #780: TestXML fails on Windows 11, Visual Studio 2022 with a Release build.

fcanderson, 2024-03-05:
[Giopestri01](https://github.com/Giopestri01) identified a unit test failure when running TestXML
(https://github.com/simbody/simbody/blob/master/SimTKcommon/tests/TestXml.cpp)

With his assistance, further investigation on his computer revealed that the source of the error is a change in the behavior of the release-build C++ Standard Libraries.  Specifically, the change in behavior occurs in ```std::ostream::setprecision(int p)```.

When run in a Debug build, a value of p = 0 results in the precision being set to 1 (the lowest precision that makes sense), and all tests pass.

When run in a Release build, however, a value of p = 0 results in the precision being set to 6, which is the default precision of std::ostream. This behavior causes the SimTK_TEST on line 396 (https://github.com/simbody/simbody/blob/master/SimTKcommon/tests/TestXml.cpp#L396) to fail for p = 0.

I don't know whether the change in std::ostream::setprecision() is due to a difference in the runtime libraries on Windows 10  vs. Windows 11, or if it's due to a very recent update of Visual Studio 2022.

Whatever the case, the fix is fairly straightforward:
Enforce ```(1 <= p <= SimTK::LosslessNumDigitsReal)``` in ```SimTK::String::String(const T& t, int p)``` before the call to ```setprecision()``` is made.

I will implement the fix and submit a PR.

sherm1, 2024-03-06:
@fcanderson can we close this issue now that #781 merged?

fcanderson, 2024-03-13:
Yes. Closing this issue.