simbody/simbody #785: Simbody core aborted while using it with gazebo classic after loading a SDF.

yaswanth1701, 2024-03-12:
System: Ubuntu 20 and Gazebo Classic.
Hi everyone, So Gazebo supports a wide range of physics engines including simbody. I am running some benchmarks with different physics engines in the gazebo and I encountered the following error.
```bash 
[Msg] Loading world file [/home/yaswanth/benchmark/build/../worlds/boxes_custom.world]
terminate called after throwing an instance of 'SimTK::Exception::Assert'
  what():  SimTK Exception thrown at State.cpp:1094:
  Internal bug detected: CacheEntryInfo::isUpToDate(): cache entry versions didn't match; current=144 stored=143.
  (Assertion 'info.getValueVersion() == m_cacheEntryVersions[i]' failed).
  Please file an Issue at https://github.com/simbody/simbody/issues.
  Include the above information and anything else needed to reproduce the problem.
Aborted (core dumped)
```

So, the benchmark involves spawning different numbers of models into the world which are loaded from SDF files. Simbody works fine up to a certain number of the model but fails once it encounters a specific number of models (eg: 21). Everything works well in case of spawning models from  C++ API but breaks once I use an SDF file and it only the case with Simbody.`ODE` and `Bullet` works as usual.

sherm1, 2024-03-12:
Thanks for the report. I checked the source and saw that the test that failed is only included in Debug builds of Simbody (which run very slowly!). It's a cache consistency check that failed. This isn't enough information for me to reproduce. Wild guess based on my vague memory of Gazebo's implementation: a threading error (missing lock or the like). It's possible that the check is overly cautious -- if you're benchmarking you should be using a Release build of Simbody anyway in which case you won't see this failure.

yaswanth1701, 2024-03-12:
Thanks a lot for the reply. I could probably send you the repo link of benchmark one once I finish the code properly. Also, I will check out the release build of simbody. Thanks again.

yaswanth1701, 2024-03-13:
Hi @sherm1, here are the benchmarks I am running for simbody 
Step to reproduce the error:
```bash 
git clone -b simbody_benchmark https://github.com/yaswanth1701/benchmark.git
cd benchmark
mkdir build && cd build 
cmake ..
make 
export GAZEBO_RESOURCE_PATH=/path/to/benchmark
. /usr/share/gazebo/setup.bash
./BENCHMARK_boxes_dt
```

closed 2024-03-13 by sherm1

yaswanth1701, 2024-03-14:
Yeah I will open a issue at OSRF. Thank you.

scpeters, 2024-03-16:
yes, please open an issue at https://github.com/scpeters/benchmark and I will follow up