simbody/simbody #793: Potential addition to class SimTK::Test

fcanderson, 2024-06-03:
Hi Sherm,

I've implemented a class (`StatesDocument`) that handles comprehensive, variable-precision de/serialization in OpenSim. It uses `SimTK::Xml` for formatting and serializes ALL states (continuous and discrete). [ Back Story: I was having trouble bringing my `SimTK::ExponentialSpringForce` class into OpenSim because OpenSim had no systematic way of gathering the auto-update discrete states that are used by `ExponentialSpringForce`. With the `StatesDocument` class now in place, things are working nicely. ]

I'm in the process of writing the unit test for `StatesDocument`. In verifying correct serialization and deserialization, I wrote a static method that might be handy to have in `SimTK::String` or, probably more appropriately, in `SimTK::Test`. The method computes an upper bound for the rounding error that can occur when converting a Real to a String at a specified precision:

```
/**
Compute the maximum error that can result from rounding a value at a
specified precision. This method assumes a base-10 representation of the value.
@param value Value to be rounded.
@param precision Number of significant figures that will be retained in value.
(1 ≤ precision ≤ SimTK::LosslessNumDigitsReal)
@return Maximum rounding error.
*/
SimTK::Real
SimTK::Test::computeMaxRoundingError(SimTK::Real value, int precision) {
    if (value == 0) return 0.0;
    int p = clamp(1, precision, SimTK::LosslessNumDigitsReal);
    SimTK::Real leastSigDigit = trunc(log10(fabs(value)) - p);
    SimTK::Real max_eps = 0.5*pow(10.0, leastSigDigit);
    if(max_eps < SimTK::LeastPositiveReal) return SimTK::LeastPositiveReal;
    return max_eps;
}
```

If you support the addition of this method, I'll add the method and enhance the appropriate unit test.

-Clay

fcanderson, 2024-06-04:
@sherm1 
I'm having second thoughts about adding the method above. I think there are better things that could be done. The method returns half the least significant digit, but there's no reason that the exact rounding error that would occur couldn't be computed. I will continue pushing things around in the sandbox and msg again if I reach better clarity on this.

sherm1, 2024-06-17:
@fcanderson FYI I've been on vacation for a month -- just catching up on emails now.