simbody/simbody #800: Trigger OpenSim CI on PR's to Simbody to catch downstream breakage?

halleysfifthinc, 2024-09-20:
Given OpenSim's status as the most prominent/heavy downstream dependent on Simbody, it might be a good idea to set up the Simbody CI to trigger OpenSim builds based on each PR to immediately catch any downstream breakage.

I believe this should be possible using [GitHub Actions reusable workflows](https://docs.github.com/en/actions/sharing-automations/reusing-workflows#calling-a-reusable-workflow), but would need some modifications to both Simbody's and OpenSim's Actions setups.

I'm opening an issue to see if there is interest by the maintainers (cc @aymanhab @sherm1 ). If so, I will try to get to this eventually, but I won't get to this for a while, in case anyone else has a more immediate interest in putting this together.

sherm1, 2024-09-23:
My thought is that rebuilding OpenSim on every Simbody PR is too much CI load. Does OpenSim build with Simbody master, Ayman? Or is it pegged to particular tags?

aymanhab, 2024-09-23:
OpenSim is pegged to a particular commit, so I agree building on every commit would be an overkill. The parallel scenario we dealt with is when API changes in opensim-core have the potential to break the GUI. We handle this by supporting a [ci build-gui] in the commit message which triggers a rebuild of the GUI based on this specific commit.

halleysfifthinc, 2024-09-26:
I recognize that OpenSim is pinned to a particular commit, but my point was that its easy for unintentional API breakage to slip into PRs. Alerting PR authors to downstream breakage before a PR is accepted should be less work in the long-term (as opposed to someone else needing to investigate potentially multiple (un)intended API changes each time the pinned Simbody commit is updated).

CI load can likely be substantially reduced (with or without testing OpenSim): Simbody CI currently doesn't include any caching, and there are a couple of different cache methods (actions@cache and/or using ccache) which could probably bring individual build times under a minute.

Manual triggering with a specific phrase (e.g. `build-opensim`) would achieve a similar goal with a reduced resource usage.