simbody/simbody #802: GazeboReactionForce tests fail on M1 MacBook

donnyward, 2024-11-01:
```
86/113 Testing: GazeboReactionForceWithAppliedForceCompliantStatic
86/113 Test: GazeboReactionForceWithAppliedForceCompliantStatic
Command: "/Users/donny/cmakebuilds/simbody-master-relwithdebinfo/GazeboReactionForceWithAppliedForceCompliantStatic"
Directory: /Users/donny/cmakebuilds/simbody-master-relwithdebinfo/Simbody/tests
"GazeboReactionForceWithAppliedForceCompliantStatic" start time: Nov 01 11:28 PDT
Output:
----------------------------------------------------------
Starting test GazeboReactionForce ...
  runOnce ...
LOW ACCURACY
Test with order 1 integator SemiExplicitEuler2, Accuracy=0.01, MaxStepSize=0.0005
t=4.9995
  joint1 a=~[-4.35472e-06,8.784e-05,1.65448e-06] w=~[-4.5361e-14,7.85653e-13,-1.85374e-06]
  joint2 qerr=0.00401764 u=-1.524e-12
  joint3 qerr=0.0025165 u=1.28084e-12
  runOnce done. real/CPU ms: 281.8 / 281.4
  Reaction 2p=~[~[-25.0009,200.882,-0.00132132],~[0.026352,0.00130637,-300]]
  Reaction 2c=~[~[25.0007,-200.882,0.101766],~[-1.23164,-0.00130637,299.997]]
  Reaction 3p=~[~[-24.9999,25.165,-0.102528],~[0.205273,0.000217728,-49.9996]]
  Reaction 3c=~[~[24.9996,-25.165,0.165439],~[-0.331097,-0.000217728,49.9989]]
Test failed due to exception: SimTK Exception thrown at GazeboReactionForceWithAppliedForceCompliant.cpp:327:
  Internal bug detected: Test values should have been numerically equivalent at tolerance=0.5.
  (Assertion 'SimTK::Test::numericallyEqual((reaction3.reactionOnParentInParent),(SpatialVec(Vec3(-25,0,0), Vec3(0,0,-50))),1,(0.5))' failed).
  Please file an Issue at https://github.com/simbody/simbody/issues.
  Include the above information and anything else needed to reproduce the problem.

GazeboReactionForce done. real/CPU ms: 283.4 / 282.9
<end of output>
Test time =   0.29 sec
----------------------------------------------------------
Test Failed.
"GazeboReactionForceWithAppliedForceCompliantStatic" end time: Nov 01 11:28 PDT
"GazeboReactionForceWithAppliedForceCompliantStatic" time elapsed: 00:00:00
----------------------------------------------------------

87/113 Testing: GazeboReactionForceWithAppliedForceRigidStatic
87/113 Test: GazeboReactionForceWithAppliedForceRigidStatic
Command: "/Users/donny/cmakebuilds/simbody-master-relwithdebinfo/GazeboReactionForceWithAppliedForceRigidStatic"
Directory: /Users/donny/cmakebuilds/simbody-master-relwithdebinfo/Simbody/tests
"GazeboReactionForceWithAppliedForceRigidStatic" start time: Nov 01 11:28 PDT
Output:
----------------------------------------------------------
Starting test GazeboReactionForce ...
  runOnce ...
LOW ACCURACY
Using acc=0.01 consTol=0.001, captureVel=0.002, minCORvel=0.002, stickVel=0.002
t=5.016
  joint1 a=~[3.28276e-19,1.09906e-18,7.62273e-18] w=~[3.28276e-16,1.09906e-15,-8.054e-17]
  joint2 qerr=0.00401729 u=-3.53307e-15
  joint3 qerr=0.00251628 u=5.58218e-16
  runOnce done. real/CPU ms: 194.3 / 194.1
  Reaction 2p=~[~[-25,200.864,3.94572e-12],~[-1.88736e-12,4.72501e-12,-300]]
  Reaction 2c=~[~[24.9998,-200.864,0.100432],~[-1.20518,-4.72501e-12,299.998]]
  Reaction 3p=~[~[-24.9998,25.1628,-0.100432],~[0.200864,1.15843e-12,-49.9996]]
  Reaction 3c=~[~[24.9995,-25.1628,0.163338],~[-0.326676,-1.15843e-12,49.9989]]
Test failed due to exception: SimTK Exception thrown at GazeboReactionForceWithAppliedForceRigid.cpp:368:
  Internal bug detected: Test values should have been numerically equivalent at tolerance=0.5.
  (Assertion 'SimTK::Test::numericallyEqual((reaction3.reactionOnParentInParent),(SpatialVec(Vec3(-25,0,0), Vec3(0,0,-50))),1,(0.5))' failed).
  Please file an Issue at https://github.com/simbody/simbody/issues.
  Include the above information and anything else needed to reproduce the problem.

GazeboReactionForce done. real/CPU ms: 195.5 / 195.4
<end of output>
Test time =   0.21 sec
----------------------------------------------------------
Test Failed.
"GazeboReactionForceWithAppliedForceRigidStatic" end time: Nov 01 11:28 PDT
"GazeboReactionForceWithAppliedForceRigidStatic" time elapsed: 00:00:00
----------------------------------------------------------
Compiled the tip of the master branch...

```

System info from fastfetch:

```
                     ..'          
                 ,xNMM.           --------------
               .OMMMMo            OS: macOS Sonoma 14.6.1 arm64
               lMM"               Host: MacBook Pro (14-inch, 2021)
     .;loddo:.  .olloddol;.       Kernel: Darwin 23.6.0
   cKMMMMMMMMMMNWMMMMMMMMMM0:     Uptime: 23 hours, 2 mins
 .KMMMMMMMMMMMMMMMMMMMMMMMWd.     Packages: 144 (brew), 1 (brew-cask)
 XMMMMMMMMMMMMMMMMMMMMMMMX.       Shell: fish 3.7.1
;MMMMMMMMMMMMMMMMMMMMMMMM:        Display (Color LCD): 3024x1964 @ 120 Hz (as 1512x982) in 14" [Built-in]
:MMMMMMMMMMMMMMMMMMMMMMMM:        DE: Aqua
.MMMMMMMMMMMMMMMMMMMMMMMMX.       WM: Quartz Compositor
 kMMMMMMMMMMMMMMMMMMMMMMMMWd.     WM Theme: Multicolor (Dark)
 'XMMMMMMMMMMMMMMMMMMMMMMMMMMk    Font: .AppleSystemUIFont [System], Helvetica [User]
  'XMMMMMMMMMMMMMMMMMMMMMMMMK.    Cursor: Fill - Black, Outline - White (32px)
    kMMMMMMMMMMMMMMMMMMMMMMd      Terminal: iTerm 3.5.5
     ;KMMMMMMMWXXWMMMMMMMk.       Terminal Font: Monaco (12pt)
       "cooc*"    "*coo'"         CPU: Apple M1 Pro (8) @ 3.23 GHz
                                  GPU: Apple M1 Pro (14) @ 1.30 GHz [Integrated]
                                  Memory: 19.16 GiB / 32.00 GiB (60%)
                                  Swap: Disabled
                                  Disk (/):
                                  Local IP 
                                  Battery (bq40z651): 
                                  Locale: en_US.UTF-8
```

sherm1, 2024-11-03:
Thanks for this report, @donnyward!

I don't have access to a Mac so will need some help. A few questions:
- is it only the low accuracy test that fails? You might have to comment out the low accuracy test to get to the high accuracy test which follows right after.
- if you build Simbody in Debug mode and run this test does it still fail?

donnyward, 2024-11-05:
Hello @sherm1,

I ran the extra tests as requested.

* I commented out the low accuracy tests, and the high accuracy tests failed as well.
* The tests still fail in Debug mode. Additionally, a few other tests are failing with the debug build:
  * TestCableSpanStatic
  * TestCableSpan_ForcesStatic
  * TestCableSpan_LengthDotStatic

Please see the attached logs and let me know if you would like me to run any other experiments.

[logs.zip](https://github.com/user-attachments/files/17634077/logs.zip)

sherm1, 2024-12-16:
cc @pepbos @adamkewley FYI some CableSpan test failures reported on Mac/Debug