simbody/simbody #805: Hard-coded visualizer search path makes binaries non (less?) relocatable

halleysfifthinc, 2024-11-25:
This is very tangentially related to #804, but the lines here

https://github.com/simbody/simbody/blob/7265c6e77bc1879e7b9602c0134ee93aa81b6900/Simbody/Visualizer/src/VisualizerProtocol.cpp#L311-L312

bake the install path into the binaries, which is not ideal.
- It leaks build system/configuration details and makes the binaries less reproducible
- It blurs the lines in CMake between configuration and install steps (CMake ideology is that the install prefix should be an install-time configurable setting, e.g.  the configure step should be completely agnostic about any future install prefix, and you should be able to set an arbitrary install prefix when running `cmake --install`)
  - Even more generally, binary tarballs should be capable of being copied/moved anywhere and still work.

The relative path between visualizer and libraries should be sufficient*,

https://github.com/simbody/simbody/blob/7265c6e77bc1879e7b9602c0134ee93aa81b6900/Simbody/Visualizer/src/VisualizerProtocol.cpp#L294-L309

but the [`Pathname::getFunctionLibraryDirectory`](https://github.com/simbody/simbody/blob/master/SimTKcommon/src/Pathname.cpp#L469) function is not implemented on Windows.

*Arbitrary install prefixes are a thing :tm: that should be generally supported, but rearranging files/directories within an installation is not something that should be allowed/accomodated.

sherm1, 2024-11-25:
Any interest in this one, @nickbianco ?

traversaro, 2024-11-25:
> but the [`Pathname::getFunctionLibraryDirectory`](https://github.com/simbody/simbody/blob/master/SimTKcommon/src/Pathname.cpp#L469) function is not implemented on Windows.

I guess you can easily implement this function in Windows as well with `GetModuleHandleEx` and `GetModuleFileNameA ` functions, see for example:
* https://github.com/ami-iit/reloc-cpp/blob/80f86d72bb9a66e54627165a31603fb6f5d6ad82/RelocCPPGenerate.cmake#L141-L156
* https://github.com/dlfcn-win32/dlfcn-win32/blob/048bff80f2bd00bb651bcc3357cb6f76e3d76fd5/src/dlfcn.c#L822-L827

nickbianco, 2024-11-25:
Currently, the visualizer is not installed next to the Simbody libraries, so I think that line could be safely removed. Happy to submit a PR for that, but I'm not sure I have cycles to dive into search paths for Windows at the moment. @traversaro's suggestion seems promising though.