simbody/simbody #808: RandomTest fails on i386

merkys, 2025-01-02:
On i386 architecture running Debian, the build of simbody v3.7 terminates with a failure in `RandomTest`. When executed manually, it says:
```
exception: SimTK Exception thrown at RandomTest.cpp:180:
  Internal bug detected: Assertion failed
  (Assertion 'value[2000] == 123.4' failed).
  Please file an Issue at https://github.com/simbody/simbody/issues.
  Include the above information and anything else needed to reproduce the problem.
```
Builds on amd64 and arm64 succeed, thus this issue seems to be confined to i386.

[Full build log](http://qa-logs.debian.net/2024/12/28/simbody_3.7+dfsg-3_unstable-i386.log)

sherm1, 2025-01-05:
Thanks for this. I suspect this test should have allowed for some roundoff when checking the value. But I don't have an i386 platform to test on. If you have a chance can you modify the test to output the actual value in `value[2000]` to see if it is within reasonable roundoff of 123.4? If so we can safely relax the test. Otherwise further investigation will be required.

merkys, 2025-01-06:
@sherm1 it prints out `123.4`, thus I guess this could be a real-valued comparison problem.

sherm1, 2025-01-06:
Thanks, that's good. If you print out `value[2000] - 123.4` it should show the amount of roundoff (I'd guess around 1e-14?). I'll update the test to use a proper tolerance if you can get me that number.

merkys, 2025-01-07:
@sherm1 good idea. I got the difference of 5.68295e-15.

merkys, 2025-01-08:
Thanks! Could you please close the issue or ping me with the commit hash when you fix it? This way I could patch the test before the next release of simbody.

sherm1, 2025-01-24:
@merkys please see PR #813

closed 2025-01-24 by sherm1