simbody/simbody #810: Release new version tag

talregev, 2025-01-21:
Release new version tag for packages managers will release it and shift it, like vcpkg and linux apt and other.

helmesjo, 2025-05-05:
Yes, a new release seem to be quite overdue. We are also planning to package Simbody for [build2](build2.org) and a tagged version is the preferred approach.

sherm1, 2025-05-05:
Agreed. I'll make a tagged release asap.

helmesjo, 2025-05-13:
Sorry for pinging, but do you have an ETA regarding this? :)

sherm1, 2025-05-13:
End of this week. I started it (see issue #821) but have a way to go.

sherm1, 2025-05-17:
OK, [Simbody 3.8](https://github.com/simbody/simbody/releases/tag/Simbody-3.8) has been released and tagged.

Please let me know if you find any problems with it. Thanks for your patience!

closed 2025-05-17 by sherm1