simbody/simbody #816: CableSpan.cpp does not compile for Debug

fcanderson, 2025-03-23:
@sherm1 @aymanhab @nickbianco 

When I set the configuration to Debug, CableSpan.cpp is not compiling. The error messages I get appear below:
```
1>CableSpan.cpp
1>C:\Users\fcand\Documents\GitHub\Simbody-source\Simbody\src\CableSpan.cpp(2521,5): error C2065: 'J': undeclared identifier
1>C:\Users\fcand\Documents\GitHub\Simbody-source\Simbody\src\CableSpan.cpp(2521,5): error C2039: 'c_NumConstraints': is not a member of '`anonymous-namespace'::MatrixWorkspace'
1>    C:\Users\fcand\Documents\GitHub\Simbody-source\Simbody\src\CableSpan.cpp(120,8):
1>    see declaration of '`anonymous-namespace'::MatrixWorkspace'
1>C:\Users\fcand\Documents\GitHub\Simbody-source\Simbody\src\CableSpan.cpp(2521,5): error C2065: 'c_NumConstraints': undeclared identifier
1>Done building project "SimTKsimbody.vcxproj" -- FAILED.
```

The code that appears to be the problem consists of ASSERTS that are being omitted when the compile is set to `Release`.

As a temporary fix, I tried commenting out those ASSERTS, but then I get an unhandled exception when I RUN_TESTS. Then, I thought, someone who really knows the code would be better at fixing it.

sherm1, 2025-03-24:
attn @pepbos 
cc @adamkewley 

Apparently we don't run Debug builds in CI! The ASSERT listed above appears to be leftover from some earlier iteration of the code. I tried some obvious changes (e.g. J to H) but replacing c_NumConstraints is trickier and I'm not sure if the same conditions are supposed to hold now.

Can you take a look when you have a chance and fix or advise?

adamkewley, 2025-03-24:
I'll try and have a look today/this week and see what the issue is!

closed 2025-05-17 by adamkewley